ThunderFly-aerospace/Hardware #4: Zálohované napájení

kaklik, 2018-11-06:
Autopilot by měl mít zálohované napájení superkondenzátory, aby bylo možné při výpadku napájení korektně ukončit záznam na SDkartu, odeslat chybové hlášení, nastavit serva do nouzové polohy a podobně. 

Zapojení bude stejné jako na modulu SUPERCAP.  S tím, že PG pin by měl být přivedený na některý pin MCU.

ChroustJan, 2018-11-07:
Jestli to dobre chapu, tak by bylo jedno PCB procesor, výstupy s konektory i napájení (měnič na 5V, ktery by mel zvládnout napájení ridici elektroniky a serv(oni jsou na 5V?) se zálohou superkondenzatiry). Senzory by byli uz zvlast?

kaklik, 2018-11-12:
napájení musí fungovat tak, že elektronika odumře v tomto pořadí: 
 
1. vypadne motor
2. Vypadne výkonové napájení
3. Vypadne napájení serv 
4. odpadne možnost autopilota něco řídit a posílat telemetrii
5. odpadne možnost autopilota vytvářet záznam

Zřejmě by to mohlo být vyřešené tak, že napájení serv bude zálohováno dvěma 50 F kondenzátory v sérii.  Pro MCU autopilota a SDkartu bude pak menší kondenzátor například. https://www.tme.eu/cz/details/bcec-5.5v-2.0f/superkondenzatory/bigcap/bce005r5c205fs/
Který zajistí, že autopilot bude moci logovat na sdkartu ještě chvíli po tom co vypadne napájení serv.