ThunderFly-aerospace/Hardware #6: Zdroj pro napájení serv  (PPM výstupu)

roman-dvorak, 2018-11-07:
5V zdroj pro napájení PPM výstup na serva přidat přímo na desku autopilota. Asi by nebylo špatné mít možnost zdroj ladit v rozsahu jednotek voltů.

Současný stav je takový, že se počítá s externím napájením cca 5V (4.5-6) z regulátoru pro motory. Přes hřebínek jsou z regulátoru napájena všechna serva.

kaklik, 2018-11-08:
[Pixhawk 4](https://docs.px4.io/en/flight_controller/pixhawk4.html) tento problém řeší tak, že mají [externí výkonový modul](https://www.banggood.com/HolyBro-PM07-Power-Management-PM-Module-w-5V-UBEC-Output-for-Pixhawk-4-PX4-Flight-Controller-p-1309166.html?cur_warehouse=CN), který reguluje napájení pro serva a autopilota a zároveň měří výkonové proudy (např do regulátorů motorů).

kaklik, 2018-11-09:
Existuje [návrh výkonového měniče](https://github.com/cernohorsky/Modules/tree/master/PowerSupply/LMR1405001A) se vstupním napětím do 40V s obvodem [LMR14050](http://www.ti.com/lit/ds/symlink/lmr14050.pdf)

kaklik, 2018-11-09:
Zajímavá možnost je řešit 5V napájení kaskádně. Tj. z bateriového napětí ~48V  stabilizovat třeba 12V a pak z 12V udělat 5V. Tam je pak na výběr z podstatně většího rozsahu obvodů.

Navíc by tím bylo umožněno, že může existovat ještě třeba dvoučlánkový zálohovací akumulátor. Pro případ, že by vypadlo napájení z hlavního pohonného akumulátoru.

kaklik, 2018-11-12:
Zatím bylo uvažováno o těchto obvodech měničů: 

- [LT3840](https://www.analog.com/media/en/technical-documentation/data-sheets/3840fa.pdf)
- LMR14050
- [LMZ35003](http://www.ti.com/lit/ds/symlink/lmz35003.pdf)
- [TPS54561](http://www.ti.com/lit/ds/symlink/tps54561-q1.pdf)
- [LM5576](http://www.ti.com/lit/ds/symlink/lm5576.pdf)

ChroustJan, 2020-12-02:
Částečným řešením může být modul: [TFSBEC01A](https://github.com/ThunderFly-aerospace/TFSBEC01)

closed 2021-05-22 by roman-dvorak