ThunderFly-aerospace/Hardware #7: Výběr senzorů do nové verze hardaware FMUv5

kaklik, 2018-11-13:
Aktuálně máme tuto tabulku, která pokrývá senzory předchozích verzí hardware autopilota ze [seznamu hardware](https://docs.px4.io/en/flight_controller/pixhawk_series.html)

| Typ senzoru        | Zaletovatelný | Má driver |  Jde koupit |
| ------------- |:-------------:|:-----:|:-----:|
| **gyroskopy, akcelerometry (IMU)** | | |
| BMI055 |  | X | X |
| ICM-20602 | | X | X |
|ICM-20689 | X | X | X |
|MPU-6000| X | X | X|
| **tlakoměry**  | |  |  |
| [MS5611](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5611-01BA03%7FB3%7Fpdf%7FEnglish%7FENG_DS_MS5611-01BA03_B3.pdf%7FCAT-BLPS0036) | X | X | X |
| **Magnetometry** |  | | |
| [IST8310](www.isentek.com/en/dlf.php?file=../ISENTEK/(201703-09)IST8310%20Datasheet%20v1.2_brief-105.09.20.pdf) |  | X |  |   |

Je tedy potřeba: 

- [ ] Najít podporovaný magnetometr, který dokážeme koupit a zeletovat
- [ ] Zjistit jak se zpracovávají data z IMU snímačů z hlediska redundance a zda je tam nějaký mechanismus, který data filtruje a kombinuje z více snímačů. Podle toho určíme, kolik IMU čidel na autopilotovi potřebujeme.

slimonslimon, 2018-11-14:
Pořadí vyhodnocování kritérií magnetometru - Driver - Pouzdro - dá se koupit

kaklik, 2018-11-20:
Tak podle toho co z kódu analyzoval @slimonslimon to vypadá, že se za provozu vyčítá pouze jeden senzor s největší prioritou a nejvyšší věrohodností (s nejnižším počtem nahlášených vlastních chyb). 
To znamená, že zde není implementován žádný chytrý algoritmus dynamické volby nejlepší měřené hodnoty, ale pouze se čeká až zvolený senzor selže. Do té doby se nic neděje.

Z toho vyplývá, že potřebujeme, aby autopilot měl 3 různé senzory s vhodným pouzdrem. A nyní alespoň 2 z nich podporované firmwarem, abychom se nepřipravili o současnou bezpečnost poskytovanou možností přepnout na záložní senzor. 

Třetí senzor by pak sloužil pro implementaci lepšího algoritmu volby měřené hodnoty v budoucnu.

slimonslimon, 2018-11-20:
ta chyba se tam počítá nějak kumulativně, když pak zase měří a nehlásí senzor chyby tak se snižuje....

kaklik, 2018-12-08:
@ChroustJan podle téhle tabulky je zřejmě už nedořešený pouze výběr magnetometru a ta nedořešenost je způsobená tím, že jsme nikdo zatím nedovedl najít magnetometr, který by splňoval výše uvedená kritéria.

kaklik, 2018-12-09:
Pro IMU  by zřejmě nejprofesionálnější řešení bylo použití něčeho, jako je tenhle modul: 
https://www.analog.com/en/products/adis16405.html

V případě současného vírníku to ale zřejmě není řešení potíží výběrem vhodného magnetometru.

kaklik, 2019-03-07:
Myslím, že magnetometr je nalezený.

closed 2019-03-07 by kaklik