ThunderFly-aerospace/Hardware #9: Telemetrický modul

kaklik, 2018-11-18:
Zřejmě by bylo rozumné vysílač telemetrie integrovat rovnou na desku autopilota, neboť autopilot bez telemetrie téměř nelze používat. 
Případný další telemetrický modul pak k autopilotu může být přípojen externě přes konektor.

kaklik, 2018-11-19:
Standardním firmware telemetrických modulů, které se používají k přenosu dat je [SiK](https://github.com/ArduPilot/SiK)
 
Který aktuálně podporuje tyto HW moduly: 

- HopeRF HM-TRP
- HopeRF RF50-DEMO
- RFD900
- RFD900u
- RFD900p

Všechny jsou založeny na těchto čipech Si
- https://www.silabs.com/products/wireless/proprietary/si10xx-sub-ghz

kaklik, 2018-11-23:
Tak zatím to vypadá na typ: https://www.silabs.com/products/wireless/proprietary/si10xx-sub-ghz/device.si1060

kaklik, 2018-11-27:
Pro potřeby pozemní stanice je zatím zřejmě nejlepší forma modemu jako MLAB modul (ISM01A), který bude součastí vícemodulové konstrukce uvnitř hliníkové krabičky. 

V hliníkové krabičce pak mohou být další podpůrné moduly (napájení, RF filtry, zesilovače, převodník na RS232 a podobně).  Z krabičky pak budou vyvedeny dva SMA konektory pro přijímací a vysílací anténu. Podobně jako je vyřešený nerozšířenějí telemetrický komunikační modul: http://ardupilot.org/copter/docs/common-rfd900.html

Oddělené antény jsou potřeba z důvodu, aby bylo možné pro přijímací a vysílací cestu zapojovat různé moduly. (Předzesilovače pro příjem a výkonové zesilovače a filtry pro vysílání) Vynecháním anténního přepínače se navíc ušetří část útlumu a zjednoduší zapojení. 

Konstrukce modemu by měla být realizovatelná pro frekvence 433MHz a 868 MHz, které leží v [ISM pásmu](https://en.wikipedia.org/wiki/ISM_band).  Změna frekvence bude pravděpodobně možná jenom změnou osazení modulů. Budou tak existovat dvě sestavy modemů v krabičkách, podle zvolené pracovní frekvence.

Antény, by pak měly být  připojeny koaxiálním kabelem a vyvedeny třeba na střechu auta, kde je možné realizovat kvalitnější anténní systém než, jednoduchou anténou našroubovanou do SMA konektoru.

Rozhodnutí které je potřeba udělat jsou:

- [ ] Nalézt zapojení pro oddělené vysílací a přijímací antény, neboť je možné že vstupní přijímací cesta bude potřebovat nějaký obvod pro blokování příjmu, aby při vysílání nedošlo k poškození přijímače.

kaklik, 2018-11-27:
Podle [této tabulky](https://www.ctu.cz/sites/default/files/obsah/ctu/vseobecne-opravneni-c.vo-r/10/12.2017-10/obrazky/vo-r10-122017-10.pdf) je vysílací výkon pro pásmo 433 MHz, pro účely telemetrie omezen na 10 mW e.i.r.p.

Pro pásmo  868 MHz je to až 500mW e.i.r.p. s klíčovacím poměrem < 10%.  V takovém případě zřejmě nedává moc smysl se snažit o konstrukci výkonového vysílacího modulu pro pásmo 433 MHz.

kaklik, 2018-11-27:
Pro výkonový zesilovač v pásmu 868  MHz by zřejmě bylo vhodné užít těchto součástek: 

- [LNA](https://cz.mouser.com/ProductDetail/Qorvo/SPF5189Z?qs=p6lVfQR1GSoxtfscJmmzLQ%3D%3D)
- [SAW Filtr](https://cz.mouser.com/ProductDetail/RF360/B39871B3715U410?qs=hkeJtpWWodCySPgmNHKkIw%3D%3D)

kaklik, 2019-03-08:
Pokusil jsem se oživit telemetrický modul.  Za tímto účelem jsem udělal tyto kroky: 

- Stáhnul aktuální [SiK firmware](https://github.com/ThunderFly-aerospace/SiK)
- Nainstaloval ubuntí balíčkovou verzi SDCC
- Zkompiloval [EC2-tools](https://github.com/SamwelOpiyo/ec2), k tomu bylo potřeba stáhnout a zkompilovat aktuální cmake, neboť balíčková verze je moc stará.
- Nainstaloval ec2tools přes checkinstall

Pak jsem se pokusil spustit libovolný nástroj z ec2-tools.  Vždy skončí chybou [Device or resource busy](https://github.com/tridge/ec2/issues/1)

roman-dvorak, 2020-11-24:
Tohle issue bylo vyřešeno, ne?

kaklik, 2020-11-25:
> Tohle issue bylo vyřešeno, ne?

Ne tak úplně. Původním smyslem tohoto issue bylo mít autopilota na jedné desce jako modem.