ThunderFly-aerospace/Hardware #16: Vytvořit tisknutelnou krabičku pro CUAV GPS NEO V2

roman-dvorak, 2019-07-17:
Nakrslit tisknutelnou krabičku pro CUAV GPS s vhodnými montážními otvory. 

Při prvním použití se tato krabička zřejmě připevní pomocí suchého zipu na podložku vírníku.

Potřebné rozměry jsou patrné z následujících obrázků. 

[image]
[image]
[image]
[image]
[image]
[image]

roman-dvorak, 2019-07-19:
Existuje základní díly krabičky pro CUAV GPS. Nyní je potřeba na krabičce vyřešit několik problémů:

- [ ] Vytvořit otvor pro GPS FIX led, která je umístěna na vrchní straně PCB. vedle antény. Její polohu lze nejlépe určit z fotek. Otvor by mohl obsahovat vytištěný světlovod. Tím dojde i k zodolnění proti vodě. 
- [ ] Vytvořit otvor pro SafetySwitch led.
- [ ] Zajistit nemožnost vytržení kabelu z konektoru. Na kabel by tam měla být nějaká svorka, která se ke krabičce přišroubuje klidně z boku
- [ ] Ověřit správnost velikosti otvorů pro šrouby

[image]

[image]

[image]

roman-dvorak, 2019-07-20:
- [ ] Použít M2.5 x 12 šrouby
- [ ] Zvětšit vnitřní poloměr části pro led o 2 mm
- [ ] Vytvořit díry pro šrouby
- [ ] Zmenšit prostor v horní části

roman-dvorak, 2019-09-26:
Do dílu je potřeba přidělat M3 montážní otvory, aby krabičku bylo možné připevnit do univerzálních drážek nového nosníku.

kaklik, 2020-10-27:
Řekl bych, že tuhle GPS zřejmě nebudeme nadále používat, tak je asi zbytečné krabičku dále vylepšovat.

closed 2020-10-27 by kaklik