ThunderFly-aerospace/Hardware #17: Zkontrolovat parametry pro M2.5

roman-dvorak, 2019-09-06:
Zkontrolovat parametry ve všech parametrických souborech pro M2.5 šrouby a matice. Nezná se mi poměr jejich hodnot. U GPS krabičky jsou nově použity M2.5 šrouby. Průměr jejich matic oproti průměru díry pro šroub se mi zdá velmi malý. Je možné, že jsou tolerance špatně nastaveny/zapomenuty

roman-dvorak, 2019-09-06:
[image]

kaklik, 2019-09-06:
Myslím, že je poměrně důležité u nových dílů přecházet na parametrickou knihovnu šroubů: https://github.com/ThunderFly-aerospace/TF-G1/tree/improved_suspension/src/lib

aby parametry šroubů nebyly v parametrickém souboru modelů/sestavy.

roman-dvorak, 2019-12-09:
Parametr je nyní správně.

closed 2019-12-09 by roman-dvorak