ThunderFly-aerospace/Hardware #19: Smart battery ze standardních akumulátorů

kaklik, 2020-05-18:
Pro akumulátor by bylo vhodné vytvořit plošný modul, který  se nalepí na stěnu akumulátoru a bude zajišťovat battery management.  Připojení k  baterii by pak mohlo být realizováno některým z konekttorů s pomocnými signálními piny. např [09442 AS120-M](https://www.tme.eu/cz/details/as120-m/konektory-dc/amass/09442-as120-m/)
Komunikace by pravděpodobně měla být vyřešena sběrnicí I²C nebo CAN.  

Tohle konstrukční řešení má pravděpodobně smysl jen u větších akumulátorů, kde se stává hmotnost a rozměry přídavné elektroniky zanedbatelná.  U menších akumulátorů je asi vhodné použít zjednodušenou verzi z issue: https://github.com/ThunderFly-aerospace/TF-G2/issues/40

kaklik, 2020-09-21:
Další typy použitelných konektorů jsou s pomocnými vývody (pro smart battery)

- https://www.tme.eu/en/details/est80_120a-mf/dc-power-connectors/
- https://www.tme.eu/cz/details/xt90i-m/konektory-dc/amass/

roman-dvorak, 2020-11-08:
https://docs.google.com/document/d/1xYBhhgjND_RS2W3Hq0Seyc-R661ge3zonk9DTIyHclc/edit#heading=h.1lxzq7rxni7v

kaklik, 2020-11-29:
PX4 firmware v současnosti podporuje [dva SMBUS obvody BQ40Z50 and BQ40Z80](https://www.ti.com/product/BQ40Z50-R2). Jsou použitelné pro 1-4 články, což je zhruba rozsah modelářských akumulátorů, které v současnosti používáme.

kaklik, 2020-12-06:
Zdrojový kód, který zajišťuje podporu pro obvody BQ40Z50 a BQ40Z80 je [tady](https://github.com/PX4/PX4-Autopilot/blob/master/src/drivers/batt_smbus/batt_smbus.h). Bohužel je implementace zřejmě nekompatibilní s obvodem [bq34z100](https://www.ti.com/lit/ds/symlink/bq34z100-g1.pdf?ts=1607244935243&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FBQ34Z100-G1), který používáme v v [LION2CELL01](https://github.com/mlab-modules/LION2CELL01)

kaklik, 2021-01-23:
V CUAV [udělali velmi podobné řešení](https://discuss.px4.io/t/smart-battery-project-from-cuav/20400).

kaklik, 2021-10-30:
Vzhledem k mizerným parametrům modelářských akumulátorů, zřejmě nedává smysl dále uvažovat o použití modelářských akumulátorů obalených nějakou pokročilejší elektronikou.  Dává ale smysl akumulátor skládat z jednotlivých článků, jak je zmiňováno v podobném issue #20.