ThunderFly-aerospace/Hardware #21: Barvy kablíků

kaklik, 2020-11-11:
Chtěl jsem vyřešit issue https://github.com/ThunderFly-aerospace/PX4UserGuide/issues/1 , ale zjistil jsem, že mám problém určit, které barvy na kablících používat. 

Pixhawk standard píše pro I2C o barvách červená, černá, černá, černá.  CUAV používá červená, bílá, žlutá, černá.  A @roman-dvorak  má u [TFASPDIMU01](https://github.com/ThunderFly-aerospace/TFASPDIMU01), červená, žlutá, zelená, černá.

roman-dvorak, 2020-11-11:
Ten můj kablík určitě vycházel z CUAV konektoru. Protože má zelený konektor (a měl na obou stranách) 

Nebral jsem tam ale vůbec ohled na standartní barvy. Nechal jsme to tak, jak to bylo. Zhledem k tomu, že není možné změnit polarita je to spíše otázkou designu. 

Osobně bych byl pro Pixhawk schéma.

Nejedná se to jen I2C ale i CAN (jen s jiným způsobem vytváření párů)

roman-dvorak, 2020-11-11:
Zubax má barevné schéma odpovídajícímu tomu, co jsi uvedl jako CUAV zapojení. (alespoň kablík jimi dodaný)

kaklik, 2020-11-11:
> Ten můj kablík určitě vycházel z CUAV konektoru. Protože má zelený konektor (a měl na obou stranách)
> 

V tom případě je vysvětlení to, že by u CUAV změnili barevný schéma. Protože kablík, který přišel s dalšími CUAV autopiloty vypadá takhle: 

[image]

> Nebral jsem tam ale vůbec ohled na standartní barvy. Nechal jsme to tak, jak to bylo. Zhledem k tomu, že není možné změnit polarita je to spíše otázkou designu.
> 
> Osobně bych byl pro Pixhawk schéma.
> 
> Nejedná se to jen I2C ale i CAN (jen s jiným způsobem vytváření párů)

V takové situaci mi přijde nejlepší použít nové CUAV/Zubax schéma pro UAVCAN. A původní CUAV/ThunderFly schéma pro I²C (Zde by ještě možná chtělo ověřit zda nedošlo k nějakému omylu a ten CUAV kablík patřil k něčemu jinýmu). 

Barevné schéma z [Pixhawk standardu](https://github.com/pixhawk/Pixhawk-Standards/blob/master/DS-009%20Pixhawk%20Connector%20Standard.pdf) se mi vůbec nelíbí. Protože tam mají všechny kablíky černo-červené, tím pádem nejde ani přibližně určit, co tam je za signály, když vidíš jen kablík.  Chápu že tohle uživetelé neřeší, ale bude to komplikovat servis.

roman-dvorak, 2020-11-11:
[image]

kaklik, 2020-11-13:
> Mám takovéto různé barevné kombinace. U toho nového kablíku od CUAV je i jiný štítek. Starší štítky vypadaly (graficky) stejně jako tady na obrázku.
> 

V téhle sadě se ale nevyskytuje vůbec ta kombinace  červená, žlutá, zelená, černá. Od které je odvozen ten kablík u TFASPDIMU01. Nicméně tahle kombinace se mi docela hodí na I2C vedení u [WINDGAUGE03](https://github.com/mlab-modules/WINDGAUGE03). Protože to otočné ložisko má na jedné polovině vývody červená-žlutá  a na druhé polovině černá zelená.  Takže je akorát stačí smotat k sobě. :)

roman-dvorak, 2020-12-09:
Mám přípravený SVG soubor se zobrazením JST-GH konektoru. Obrázek není dodělaný; aktuální stav je vidět tady. Samtoný SVG soubor je níže zabalený v ZIP souboru. 
[image]
[jstgh.zip](https://github.com/ThunderFly-aerospace/Hardware/files/5666047/jstgh.zip)

kaklik, 2022-01-13:
Přesunuto a publikováno v [oficiální PX4 dokumentaci](https://docs.px4.io/master/en/assembly/cable_wiring.html).

Mergnutím pull-requestu https://github.com/PX4/PX4-user_guide/pull/1724#event-5884243095

closed 2022-01-13 by kaklik

roman-dvorak, 2022-03-20:
Toto barevné schéma má trošku nepříjemnost v záležitosti, kdy by uživatel chtěl použít UART port na CUAV autopilotech, který obsahuje UART a I2C. 

[image]

Pokud zde použije ThunderFly UART kablík a/nebo z něj vytvoří rozdvojku UART/I2C, tak v I2C konektoru budou pro SDA/SCL barvy modrá/žlutá. 

Podle I2C barev by tam měla být žlutá/zelená. 

**I2C**
[image]

**UART**
[image]

Ale v konektoru GPS/SAFETY kde je pořadí pinů UART, I2C je očekávané barevné schéma (podle dílčích barevných značení)
[image]

### Proto bych u UART/TELEM pinoutu očekával barvy:

|Signal | ThunderFly color |
|--------|------------------------|
| +5V    |  [image]  Red    |
| TX  |   [image] White | 
| RX |  [image] Green  | 
| CTS/SCL  |     [image]  Yellow  | 
| RTS/SDA |       [image] Green  | 
| GND |   [image]  Black |