ThunderFly-aerospace/Hardware #23: I2C extender

kaklik, 2021-03-29:
V Pixhawk systému zřejmě chybí nástroj, jak prodlužovat vodiče I2C sběrnice. Zřejmě by to měl být modul, který na protilehlých hranách bude mít běžné JST-GH konektory používané pro I2C. 

Funkci modulu lze odvodit od existujících extenderů používaných pro jiné aplikace:

- https://www.poscope.com/product/poneti2cextender/
- https://blog.adafruit.com/2020/10/14/new-guide-adafruit-ltc4311-i2c-extender-active-terminator-adafruitlearningsystem-stemma-adafruit/
- https://store.ncd.io/product/p82b715td-long-distance-i2c-bus-extender-i2c-mini-module/

ChroustJan, 2021-03-29:
@kaklik jako zásadní vidím výběr vhodného IO. A pak mechanickou konstrukci. Očekávám, že to budeme chtít malé.

kaklik, 2021-04-03:
Zvolen obvod [TCA4307](https://www.ti.com/product/TCA4307)

closed 2021-04-03 by kaklik