ThunderFly-aerospace/Hardware #24: RC vysílačky

roman-dvorak, 2021-05-06:
Potřebujeme vybrat vhodnou vysílačku pro používání s UAV. 

Aktuálně máme velké zkušenosti s: FRsky [taranis x9d plus](https://www.frsky-rc.com/product/taranis-x9d-plus-2/)
A menší s Jumper T16. 

Nevýhody obou vysílaček jsou:
 * Velmi nekompaktní, velké do ruky. Pokud člověk stojí/sedí, je to OK. Ale v případě, že s tím má vystupovat z auta atd.. tak je to dost nekomfortní. A pak se hodně bojí, aby nepohnul s kniply např o oblečení. 
 * Nekvalitní provedení, plastový obal
 * Nekvalitní páčky - u x9d se povedlo neznámým způsobem zlomit jeden přepínač. Naštěstí lze vysílačky snadno rozebrat a vypínač vyměnit za jiný. Je to nějaký generický páčkový vypínač
 * Kniply mají odporový senzor polohy - není moc přesný a otázka je, co se stane s vodou/vlhkostí. Lze je nahradit za hallovy senzory. 

X9D se nenabíjí, když je spuštěná a T16 nemá nabíjecí konektor. Ale obsahuje 2 Liion články (18650). 

Jumper T16, má řadu tlačítek, které lze namapovat na jeden kanál. Na tomto kanálu pak lze mít snadno několik letových režimů zároveň. Je to zároveň lépe vidět, takže si operátor tak moc nemusí dávat pozor na přepínač, o který může zavadit. T16 má ještě velký barevný LCD display (který je relativně čitelný na slunci). Lze na něj nahrát některé LUA skripty. Do kterých aktuálně neumíme dostat data z důvodu nekompatibility PX4. Ale neukazuje se to jako kritická věc pro aktuální režim létání. Asi to bude důležité pro daleké a near-BVLOS lety. 

Vysílačka FRsky podporuje aktuálně jen ACCESS komunikační protokol. FRsky nově používá ACCST.
Jumper T16 není od FRsky a podporu pro ACCESS má vyřešenou multiprotokolovým modulem, který podporuje mimo jiné i ACCESS (nevím, jsetli podporuje ACCST nebo ne??). Zatím nebyl pozorován žádný problém v nekompatibilitě. 

Vzhledem k tomu, že aktuálně se jedná o dvě vysílačky a jednotky přijímačů, tak dává smyl přejít na novější protokol ACCST. 

Novinkou u FRsky je tzv. Fport, což je nějaká komunikační linka, která se připojí mezi RC přijímačem a FC pro přenos telemetrických dat. Má to nahradit aktuální S.Bus a S.Port. Nevím, jakou to má kompatibilitu s PX4. 

Ve všech připadech je dost nevyhovující řešení napájení a nabíjení. 

Po průzkumu existujících řešení jsou vyhovující

Taranis X9D - https://www.frsky-rc.com/product/taranis-q-x7s/
* Proprietární nabíječka
* Hallovy kniply

Taranis X9D Plus SE 2019 - https://www.frsky-rc.com/product/taranis-x9d-plus-se-2019/
* Novější verze X9D
* Hallovy senzory
* RAM 192K

T16 pro hall - https://www.jumper-rc.com/products/transmitters/t16-pro-hall/
* Hallovy senzory
* RAM 8MB

kaklik, 2021-05-06:
@roman-dvorak  Dobře, že jsi prošel nové možnosti. :)   Když si to sepisoval, vycházel si i z  [OpenTX seznamu](https://www.open-tx.org/radios.html)? Mě v něm totiž zaujala ještě [Radiomaster TX16S](https://www.radiomasterrc.com/articleb-113.html), která mi kromě běžných vlastností, které si hodnotil přijde zajímavá ještě tím, že má dva UART porty přístupné z interních skriptů a dobíjí se přes USB-C a uvnitř zřejmě jsou 18650 akumulátory. :)

roman-dvorak, 2021-05-06:
Ano, na ten seznam jsem se také díval. A zaměňoval jsem to za tu Jumper T16 - Je dost zajímavé, jaká je v nich podobnost (i v jiných modelech Jumper a Radiomaster)  - připsal jsem ji do seznamu. 

To nabíjení je dost velký benefit. Předpokládám, že to zase má nějakou multiprotokol RF část. Zatím jsme na nekompatibilitu nenarazili. Ale je otázka, jak to bude do budoucna, kdy FRsky zavádí nové vlastnosti (ACCST, Fport, ...)

kaklik, 2021-05-06:
> Ano, na ten seznam jsem se také díval. A zaměňoval jsem to za tu Jumper T16 - Je dost zajímavé, jaká je v nich podobnost (i v jiných modelech Jumper a Radiomaster) - připsal jsem ji do seznamu.
> 

Ta podobnost zřejmě vychází z toho, že na Radiomaster zřejmě pracovali lidi, [kteří původně navrhovali tu vysílačku pro Jumper](https://www.radiomaster.nl/radiomaster-tx16s-vs-jumper-t16-pro-v2/).

> To nabíjení je dost velký benefit. Předpokládám, že to zase má nějakou multiprotokol RF část. Zatím jsme na nekompatibilitu nenarazili. Ale je otázka, jak to bude do budoucna, kdy FRsky zavádí nové vlastnosti (ACCST, Fport, ...)
>

Podle [mého odkazu výše](https://www.radiomaster.nl/radiomaster-tx16s-vs-jumper-t16-pro-v2/), to USB nabíjení má zřejmě i Jumper.
Myslím si, že Frsky se zaváděním těch nových vlastností pouze udržuje pozici na trhu. Vzniká tak otázka, zda neexistuje i nějaký univerzální přijímač? (Třeba s open-source firmwarem)  Protože to co používá Frsky jsou stejně jen siliconlabs čipy s nějakým Frsky firmwarem který řeší protokol a kódování. 
 
> Společný problém pro všechny vysílačky je velmi špatná odolnost proti vodě/počasi.

Jasně no. To by ale vyřešila pouze úplně jiná konstrukce vysílačky. Ve vývojovém plánu jsem například našel poznamenáný [Micronav](https://www.uxvtechnologies.com/ground-control-stations/micronav/?uavref=1).

roman-dvorak, 2021-05-06:
> Jasně no. To by ale vyřešila pouze úplně jiná konstrukce vysílačky. Ve vývojovém plánu jsem například našel poznamenáný Micronav.

Takových designů existuje více (Auterion, CUAV, ...). Možná by stálo za to uvažovat o tom něco takového experimentálně pořídit. Ale asi to má trochu jiné užití než tyto RC vysílačky. Ale pro měření pod mraky (a BVLOS) by to bylo asi docela šikovné.

roman-dvorak, 2021-05-06:
Já jsem původně hledal jen náhradu za tu RC vysílačku, co použiváme.

Pak vím například o tomto: http://www.os-rc.com/en/, kde se asi dost snaží vyřešit některé problémy dnešních vysílaček.

roman-dvorak, 2021-05-07:
https://www.peckamodel.cz/tx16s-vysilac-tx16s-radiomaster

kaklik, 2024-05-05:
Aktuálně vidím jako nejlepší řešení, přímé řízení do UAV zrušit a mít jen jednu spolehlivější datovou linku pro [C2](https://en.wikipedia.org/wiki/Command_and_control)

closed 2024-05-05 by kaklik