ThunderFly-aerospace/Hardware #26: Autogyro rotorhead PCB

kaklik, 2022-01-28:
Autogyro has specific requirements of control equipment mounted in the rotor head. Therefore a combined PCB with the following functions would be useful:

- CAN bus connection (UAVCAN)
- Two PWM outputs for rotorhead servos
- Rotor bearing temperature measurement
- Rotor RPM sensing
- Airspeed sensor connection
- Vibration sensing
- Optional external magnetometer (rotor head magnetometer mount is advantageous due to minimal RF interference from autogyro onboard instruments)
- Prerotator control  (Low power, high torque prerotator motor control)
- Backup power supply option from rotor energy generated by prerotator motor
- Middle power connection to onboard avionics, to power prerotator engine, or power backup for autopilot and servos in case of main power supply failure. 

All above functions could be either combined to single PCB or multiple PCBs which could be combined modularly.

roman-dvorak, 2022-12-03:
Jako zajímavý základ pro tuto desku vypadá tohle: https://arkelectron.com/product/ark-cannode/ Jsou k tomu dostupná schémata, přpravený bootloader pro PX4 i samotná PX4 binárka. Vývoj s tímto zařízením by tedy běl být značně usnadněn. 

Přípojuje se to přes CAN k autopilotu. Díky použítí PX4 bootloaderu to například podporuje aktualizaci FW skrze CAN.

roman-dvorak, 2022-12-03:
### Měření RPM

PCF8583 má relativně velké pouzdro. 

Pokud by mělo být měření zpřesněno, tak by bylo asi zajímavé na čítací vstup umístit nějaký vypínačí prvek (schmidtúv KO s EN vstupem, nějaké AND nebo tranzistor) tak, aby procesor mohl v definovaný čas zastavit čítání, vyčíst hodnotu a následně zase v definovaný čas čítání zapnout. Tím by se měla zmenšit rozptyl měřených hodnot, protože to nebude záviset na tom, jak v nuttx vyjde komunikace přes I2C. Díky tomu bychom mohli mít hodnoty o něco častěji se stejnou přesností. 

Jako alternativou by mohl být [S-35770E01I-K8T2U](https://cz.mouser.com/datasheet/2/360/S35770_I_E-1628617.pdf), který je zřejmě lepší v následujícím: 
 - Pří napájení nad 2v5 umí čítat do 1MHz (PCF8583 to nemá definované, experimentálně (s generátorem) vyzkoušeno 22kHz
 - Obsahuje integrovaný Schmidtův KO, zároveň to má definovaný maximální raise/fall time, takže i tak by bylo vhodné to vybavit externím KO. Možná v kombinaci s "vypínačem měření" (viz výše)
 - Má to 24 bit čítač, data jsou uložena v binární podobě ve trojici registrů - není nutné používat všechny (PCF8583 čítá v BCD kódu ve 4 registrech, tím by se mělo urychlit vyčítání)
 - Má to pin pro reset - což by se využilo k urychlení vyčítání v případě integrace na jednu desku a hlavně přesné definování času začátku čítání. 
 - Menší pouzdro

Myslím, že pro optimální funkci je důležité co nejvíce urychlit vyčítací proces a přesně nadefinovat časy začátku a konce měření